RF1000/Repetier-Firmware #8: OUTPUT_OBJECT_SCRIPT Logically Incorrect GCode - v 0.91.51 - Can cause CRASH

rf1k-mjh11, 2015-02-11:
I actually prepared the whole text beforehand in German.  So see further below for the issue in German.  Here it is in English:

The printer allows outputting the object either via menu option & button on the printer itself, or via gcode command M3079.  I assume one or both of these end up calling the script shown in "Configuration.h", line 1503.

[i]#define	OUTPUT_OBJECT_SCRIPT	"G21\nG91\nG1 E-10\nG1 Z210 F5000\nG1 X0 Y220 F7500\n"[/i]

That script switches to relative coordinates (via G91) to be able to retract the filament 10mm.  Then the script proceeds to send positioning commands in X, Y and Z.  But since the printer is still in relative mode, these commands do not achieve what is intended.
Here's the script, plus my comments:

G21	; units are in "mm"
G91	; switch to relative coordinate entry
G1 E-10	; retract 10mm of filament from current position
G1 Z210 F5000	; move bed from current position 210mm down (could cause a crash, in theory, see [b]note[/b] further below).
G1 X0 Y220 F7500	; move from current position 0mm in X and 220mm in Y direction (does cause a crash against a hard stop in Y).

[b]NOTE:[/b] I verified the crash in Y, but there is no crash in Z direction, I assume there is another firmware safeguard in place that prohibits that.

I have two solutions available:
One:
#define	OUTPUT_OBJECT_SCRIPT	"G21\nG91\nG1 E-10\nG90\nG1 Z210 F5000\nG1 X0 Y220 F7500\n"
Here we switch back to absolute coordinate mode (via G90) right after retracting the filament, so the intended coordinates in X, Y and Z are properly targeted.

Or two:
#define	OUTPUT_OBJECT_SCRIPT	"G21\nM83\nG1 E-10\nG1 Z210 F5000\nG1 X0 Y220 F7500\n"
Here we only switch the extruder to relative mode (via M83) while everything else remains in absolute coordinate mode.  The extruder retracts as planned, and the proper coordinates are targeted without any chance of a crash (provided no steps were lost).

I verified that the script causes a crash in Y by manually inputting the script commands in Repetier-host.  I can no longer verify if both the printer/button approach and the M3079 command both cause the issue, since I changed the firmware before uploading (using version 2), so the script does not cause any problems for me.

mjh11

Now for the German:
===========================================================================
Kleines Problem in Firmware 0.91.51

Die Firmware bietet eine eigene Routine um das (fertige) Teil "auszugeben".  Gemeint ist damit, dass die Zugänglichkeit zum gedruckten Teil, durch das Herunter- Vorfahren des Betts sowie das ganz nach links Fahren des Extruders, erhöht wird.  Dazu wird in der Configuration.H ein Skript definiert das entsprechend ausgeführt wird.

In der Zeile 1503:
#define	OUTPUT_OBJECT_SCRIPT	"G21\nG91\nG1 E-10\nG1 Z210 F5000\nG1 X0 Y220 F7500\n"

Im Klartext:
G21	; Werte sind in "mm"
G91	; wechsle in den relativen Modus (Bewegungen werden relativ)
G1 E-10	; ziehe von der augenblicklichen Position 10mm Filament wieder ein (da wir im relativen Modus sind)
G1 Z210 F5000	; fahre von der augenblicklichen Position noch 210mm tiefer (ist da ein Crash vorprogramiert?)
G1 X0 Y220 F7500	; fahre von der augenblicklichen Position noch 0mm in X-Richtung und 220mm in Y-Richtung (ist da ein Crash in Y nicht vorprogramiert?)

Zum Glück würden die Crashes nicht zu einem Düsen/Extruderschaden, oder zu einer gebrochenen Keramikplatte führen, sondern wahrscheinlich nur lautes Motor-Rattern hervorrufen, da auf Block 
gefahren werden würde. Der erste Teil des Skripts, "G1 X0" findet im relativen Modus statt, folglich ist es ziemlich harmlos (es findet keine Bewegung statt).  Der zweite Teil, Y220 verlangt aber, dass das beheizte Bett von der aktuellen Position 220mm nach Vorne fährt.  Da in 99.9% der Fälle das Bett nicht in der hintersten Position sein wird, fährt das Bett mechanisch auf Anschlag.  Das habe ich mittels manueller Eingabe der Befehle nachvollzogen.  Interessanterweise fährt aber das Bett nicht auf einen mechanischen Anschlag nach unten, sondern bleibt rechtzeitig stehen.  Wahrscheinlich handelt es sich in diesem Fall um eine zusätzliche Firmware-Absicherung, die dieses verhindert.
Ein kleines Manko ist hier, dass der Extruder eben an derselben Position bleibt, und nicht nach links fährt.  Es entsteht dadurch aber kein Schaden.

Als Lösung biete ich zwei Alternativen an:

Version 'A'
#define	OUTPUT_OBJECT_SCRIPT	"G21\nG91\nG1 E-10\nG90\nG1 Z210 F5000\nG1 X0 Y220 F7500\n"

Hier kommt nach dem relativem Modus (G91), der für den Retract des Filaments notwendig ist, gleich wieder der absolute Modus ("G90"), damit die X, Y und Z Koordinaten wieder absolut werden.  Es sollte kein Crash vorkommen, vorausgesetzt, die Schrittmotoren haben keine Schritte verloren.  Das Bett fährt ganz hinunter und nach vorne, und der Extruder ganz nach links.  HINWEIS:  Für die Z-Koordinate wird 210mm angefahren, in der Firmware (Zeile 755) wird als Z_MAX_LENGTH 200 angegeben.  Trotzdem kann man auf 210mm fahren.

Version 'B'
#define	OUTPUT_OBJECT_SCRIPT	"G21\nM83\nG1 E-10\nG1 Z210 F5000\nG1 X0 Y220 F7500\n"

Hier wird statt der relativen Positionierung (G91) die absolute beibehalten, stattdessen wird M83 verwendet um nur den Extruder in den relativen Modus zu schalten.  Damit funktioniert der Retract wie erwartet, und das Bett fährt ebenso ganz hinunter und nach vorne, mit dem Extruder ganz links.

mjh11

RF1000, 2015-02-11:
Hello,

first of all, you are right - calling of "Output Object" always runs the G-Codes from the OUTPUT_OBJECT_SCRIPT.
The rest is not so clear - even though we use relative coordinates here, the printer should not move to any position which is outside the physically "healthy" limits. Moving 220 mm relatively means that the heat bed is moved UP TO 220 mm - and in case the max y-position is reached before the full way is moved, the heat bed is stopped earlier (and there should not be any kind of crash).
This is the same for all directions.

What happens exactly when you get the "crash" in y-direction with the original OUTPUT_OBJECT_SCRIPT? This output-object is part of the standard RF1000 configuration so it should be used by a wide range of people (and is used during our tests) and we did not notice a misbehavior of the movement in y-direction so far.

Best Regards,
    RF1000

shadow974, 2015-02-12:
Die Halterung des Zahnriemens knallt gegen das vordere Kugellager. Teilweise rattert der Schrittmotor dann noch etwas weiter weil es den Soll-Punkt noch nicht erreicht hat.

RF1000, 2015-02-12:
Kann man vielleicht die exakten Schritte beschreiben, die ab dem Einschalten bis zum Auftreten des Fehlers durchzuführen sind?
Theoretisch könnte die maximale y-Länge überfahren werden wenn Output Object zu einem Zeitpunkt ausgeführt wird, wo die y-Home Position nicht bekannt ist (in dem Fall weiß der RF1000 schlicht nicht, auf welcher y-Position er gerade ist und kann dementsprechend über die maximale Länge hinaus fahren). Da Output Object aber nur die G-Codes abarbeitet wenn die Home-Position bekannt ist müsste jemand nur die x- oder nur die z-Achse gehomt haben, um so ein Verhalten "zu erreichen" (die Firmware hat bisher nur ein globales "Home Position bekannt/nicht bekannt", keines pro Achse). Bei einem normalen Druck-/Fräsvorgang sollte das daher nicht auftreten (bei manuellem Senden von G-Codes aber evtl. schon).

shadow974, 2015-02-12:
Bei mir ist das jedes mal passiert, wenn ein Druck abgeschlossen wurde. Also war da definitiv ein Homing davor. Abhängig davon, wie weit "hinten" der Druck beendet wurde, ist er dann dagegen geknallt, oder nicht.

RF1000, 2015-02-12:
Also mit "G1 Y220" müsste es egal sein, wie weit vorne oder hinten du gedruckt hast ... von dort weg + 220m müsste eigentlich so gut immer die maximale y-Position erreichen (Y_MAX_LENGTH = 245).
Hast du einen beispielhaften (idealerweise kurzen) G-Code, mit dem wir das nachstellen können?

shadow974, 2015-02-12:
Vielleicht ist "245" einfach zu viel? Oder de Begrenzung greift nicht. Ich habe mir leider nicht gemerkt, welcher meiner gcodes das das letzte mal verursacht hat :(

rf1k-mjh11, 2015-02-12:
Shadow974 & RF1000,

Ich muss vielleicht zugestehen dass Y bloß bei manchen Drucker zu einem 
Crash führt.  Ich habe mein Y-Endstop so eingestellt, dass die Düse bloß 
3mm über die Keramikplatte vorsteht um mehr Druckfläche zu erreichen.   
Der Endanschlag ist somit nicht so eingestellt wie es in der 
Montage-/Betriebsanleitung vorgesehen war.  Dadurch ist mein Verfahrweg 
in Y nicht wie vorgesehen.  Könnte das ein Grund sein, dass es bei mir 
auf Anschlag fährt und rattert?

Die genauen Schritte die ich gemacht habe, waren (zur Klarstellung: ich 
hatte meine Firmware schon geändert, bevor ich diese Tests machte, daher 
musste ich die GCodes manuell über Repetier-host eingeben):

Im Host wurde der "Home All"-Button gedrückt (Drucker fährt nacheinander 
alle Home-Positionen an)
Dann per manueller GCode-Eingabe, diverse Punkte angefahren (simulierter 
Druck), z.B.:

G1 X95 Y 105 Z20 F3000    ; gefolgt von ähnlichen Codes, mit steigendem 
Z, z.B.:
G1 X50 Y155 Z34
G1 X110.5 Y60 Z123.5

Danach habe ich die Codes, wie sie im Skript stehen, nacheinander 
eingegeben.
Der Erwartete Crash in Z, da er ja 333.5mm anvisiert (210+123.5), kam 
nicht, da greift offensichtlich eine weitere Firmware-Absicherung.
In X tut sich nichts, da ja eine relative Bewegung von 0 keine Bewegung 
hervorruft (das allein ist schon ein Hinweis auf das Vorliegen eines 
Gedankenfehlers).
In Y ergab sich jeweils ein Fahren bis auf einen mechanischen Anschlag 
mit dem verbundenen (sehr) kurzzeitigen rattern.  Und da hat RF1000 
wahrscheinlich recht, dass eine Firmware-Absicherung auch am eingreifen 
ist, da das Rattern so kurz ist.  Nachdem, bei einer relativen Bewegung, 
die Y-Achse das Ziel "270mm" (210+60) anvisiert, müsste das Rattern doch 
einige Zehntel-sekunden dauern.

Ich mache noch einen Versuch, wo ich versuchen werde ein "unheimlich 
langes" Rattern zu provozieren indem ich von einem Y-Wert um die 200 
herum ausgehe.  Da sollte ja, im relativen Modus, ein Y-Wert von ca. 300 
angepeilt werden.  Wenn das Rattern sehr kurz ist, greift garantiert 
irgendeine weitere Firmware-Routine ein.  Damit könnte ich annehmen, 
dass meine Abweichende Y-Anordnung zum Crash beiträgt.

Ich steige in Kürze ins GitHub ein und werde meine Aussage entsprechend 
updaten.

mjh11

RF1000, 2015-02-12:
Hallo,

zu "Könnte das ein Grund sein, dass es bei mir auf Anschlag fährt und rattert?":
-> Natürlich, wenn dein Bett in Y-Richtung z.B. nicht mehr 245 mm fahren kann bis es rattert, dann rattert es jetzt bei dir :)

zu "(das allein ist schon ein Hinweis auf das Vorliegen eines Gedankenfehlers)":
-> Gedankenfehler von wem? Wie beschrieben sollte eine relative Bewegung in Y-Richtung von 220 mm (oder auch von 500 mm) niemals einen Crash verursachen, weil die Firmware ja "weiß" wie weit sie in Y-Richtung fahren kann bevor das Ende (= der Software Y-Max Endstopp) erreicht worden ist. 

Wenn deine veränderte Y-Konfiguration der Grund für dein Rattern ist dann kannst du einfach einmal ausmessen, wie weit das Heizbett vom Z-Min Endschalter weg bewegt werden kann bis es zum Rattern kommt ... und einen etwas kleineren Wert solltest du dann für Y_MAX_LENGTH eintragen. Damit sollte dann dein Rattern weg sein.
Bleibt noch das vom shadow ...

rf1k-mjh11, 2015-02-12:
RF1000,

Deine Aussage scheint zu stimmen, dass der Output Skript nur bei 
bekannter Position funktioniert.  Gestern hatte ich einmal, unmittelbar 
nach der erstmaligen manuellen Eingabe der Skript-GCodes auch versucht, 
das M3079 Kommando abzugeben.  Da geschah nichts, und im Log stand etwas 
wie "can't do it, position unknown", was mich einen Augenblick wunderte, 
da ich dachte, der Drucker müsste doch genau wissen wo er ist.  Dann 
fiel mir im Repetier-host auf, dass der Z-Wert eben auf den (vorhin 
mittels simulierten Skript) erzwungenen Wert von 235.5 stand, und der 
Wert eben in rot dargestellt war.  Hier hat also die Firmwareabsicherung 
gegriffen und anschließend erkannt, dass die Z-Position unbekannt sei, oder?

mjh11

RF1000, 2015-02-12:
Ja, Output Object legt nur los, wenn der Drucker seine Position kennt (was nur dann der Fall ist, wenn a) ein Homing durchgeführt worden ist und b) die Stepper danach nicht ausgeschalten worden sind).
Wenn der Drucker seine Position nicht kennt und einfach einmal um 220 mm in eine Richtung fahren würde könnte er tatsächlich versuchen, über die mechanische Grenze drüberzufahren (was man nicht will).

shadow974, 2015-02-12:
Der Y-Referenzschalter ist ein sehr guter Punkt. Warum habe ich selbst nicht daran gedacht!? Ich werde das heute nochmal prüfen und dann einfach den Referenzpunkt nachjustieren.

rf1k-mjh11, 2015-02-12:
RF1000,

Zum Rattern:

Korrekt, mein Verfahrweg ist (gerade frisch vermessen) nur 232mm.
Ich dachte damals, ich bekomme mehr Verfahrweg wenn ich die Düse nur 
kurz überstehen lasse.  Das wird sofort "zurückgebaut"!  Damit wird die 
Zugänglichkeit zur Düse natürlich verbessert (bisher bin ich einfach auf 
Z=10 gefahren beim Putzen).

Zu Gedankenfehler:

Die Tatsache, dass, im relativen Modus, ein Befehl wie "G1 X0" 
eingegeben wird, deutet auf den Gedankenfehler hin.  Ist man im 
relativen Modus, führt der Befehl zu keiner Bewegung.  Also bleibt der 
Extruder an seine Position.  Offensichtlich will man, dass der Extruder 
ganz nach Links fährt, um die Entnahme des Werkstücks zu erleichtern.
Falls die X-Achse auch mittels Firmware-Routine gegen Überfahren 
abgesichert ist, müsste man statt "X0" etwas wie "X-220" eingeben, damit 
der Wagen ganz nach Links fährt.  Das wäre natürlich auch eine Möglichkeit.

Aber was spricht eigentlich gegen die zwei von mir angebotenen 
Lösungen?  Da wird der Absolutmodus nie, oder nur kurz verlassen, und 
alle Koordinatenangaben bleiben verständlich.

mjh11

RF1000, 2015-02-12:
Hallo,

du hast recht, G1 X0 im relativen Modus bringt nichts und wurde auch bereits aus dem OUTPUT_OBJECT_SCRIPT entfernt (was für euch beim nächsten Update auf Github sichtbar wird).

Ob man (immer) will dass der Extruder ganz nach links fährt oder nicht weiß ich nicht - im Grunde kommt man besser an den Extruder ran wenn er irgendwo in der Mitte bleibt. Nach Output Object sind aber die Stepper sowieso aus und man kann den Extruder beliebig mit der Hand hin und her bewegen.
Ganz nach links fahren würde in x-Richtung übrigens im X-Min Endschalter enden, und dort sollte es daher überhaupt keine Chance für ein Rattern geben (weil das ja ein echter Endschalter ist, der hoffentlich so verbaut ist dass der Endschalter auslöst bevor irgend etwas anderes ansteht).

Im Grunde spricht nichts gegen deine beiden Alternativen - die Idee von Output Object war einfach, dass der Tisch ganz nach unten und ganz raus fährt. Ich denke das wird sowohl mit der aktuellen als auch mit deinen Lösungen erreicht. Und auch deine Lösung (mit absoluten Koordinaten) würde rattern, wenn der Wert von Y_MAX_LENGTH nicht mit der tatsächlich zur Verfügung stehenden y-Länge übereinstimmen würde :)

rf1k-mjh11, 2015-02-12:
RF1000,

Bin gerade im RF1000-Forum und schreibe einen Update.

Ich stimme bei, das ich gerne den Extruder mehr in der Mitte habe. Ich 
verwende eigentlich den OUTPUT_SCRIPT gar nicht.
Wurde nun nur der "X0"-Teil rausgenommen, oder auf etwas wie "X-200" 
geändert?  Ach, das kann ich selber nachgucken.  Ich steige bald ins 
GitHub ein um mein Issue als CLOSED zu markieren (mit Erklärung).

Extruder selbst verschieben (tue ich immer):
Der Skript enthält aber keinen Befehl, die Motore auszuschalten (wäre 
M84, glaube ich).  Das geschieht wahrscheinlich anders wo in der Firmware.

mjh11

RF1000, 2015-02-12:
Das neue OUTPUT_OBJECT_SCRIPT wird "G21\nG91\nG1 E-10\nG1 Z210 F5000\nG1 Y250 F7500\n" sein, sprich das "X0" verschwindet und der Y-Wert wird sogar "noch unmöglicher" (was aber für den Zweck der Funktion irrelevant ist).

Und im Script ist M84 nicht notwendig weil Output Object per Definition und von sich aus nach der Abarbeitung von OUTPUT_OBJECT_SCRIPT die Stepper ausschaltet.

rf1k-mjh11, 2015-02-12:
Concluding statement from Issue originator:

After resetting the Y-axis endstop to the originally designed position, no crash in Y was experienced when simulating the script.
The (non-effective) X0 command (in relative mode) has been (or will be) removed from the script per the previous comment by RF1000.
As a result, I consider the issue as resolved.

GERMAN:
Nachdem der Anschlag für den Y-Endschalter in die Position laut Montageanleitung gebracht wurde, führte das OUTPUT_OBJECT_SCRIPT nicht mehr zu einem Fahren bis auf Anschlag.  Der wirkungslose X0 Befehl im Skript wird (oder wurde) laut verhergehendem Kommentar von RF1000 aus dem Skript entfernt.
Basierend darauf erachte ich diesen Punkt als gelöst.

closed 2015-02-12 by rf1k-mjh11