RF1000/Repetier-Firmware #22: RF1.37 do not build

GaramMasala, 2017-04-12:
Hello!

Rf1.37 Do not build with arduino 1.6.12 But it does with arduino 1.0.6
Here is the build log from 1.6.12

Arduino: 1.6.12 (FreeBSD), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

/usr/local/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/local/arduino/hardware -tools /usr/local/arduino/tools-builder -built-in-libraries /usr/local/arduino/libraries -libraries /usr/home/bernt/sketchbook/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10612 -build-path /tmp/arduino_build_147243 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /usr/home/bernt/nerladdningar/vebb/Repetier-Firmware-development/RF1000/Repetier/Repetier.ino
/usr/local/arduino/arduino-builder -compile -logger=machine -hardware /usr/local/arduino/hardware -tools /usr/local/arduino/tools-builder -built-in-libraries /usr/local/arduino/libraries -libraries /usr/home/bernt/sketchbook/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10612 -build-path /tmp/arduino_build_147243 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /usr/home/bernt/nerladdningar/vebb/Repetier-Firmware-development/RF1000/Repetier/Repetier.ino
Using board 'mega' from platform in folder: /usr/local/arduino/hardware/arduino/avr
Using core 'arduino' from platform in folder: /usr/local/arduino/hardware/arduino/avr
Detecting libraries used...
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_147243/sketch/Repetier.ino.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "/tmp/arduino_build_147243/sketch/Repetier.ino.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/Repetier.ino.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/Commands.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/Communication.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/Eeprom.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/Extruder.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/HAL.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/Printer.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/RF.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr/local/arduino/hardware/arduino/avr/libraries/Wire/src" "/tmp/arduino_build_147243/sketch/SDCard.cpp" -o "/dev/null"
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/mega" "-I/usr/local/arduino/hardware/arduino/avr/libraries/SPI/src" "-I/usr [... cut ...]
/tmp/arduino_build_147243/sketch/pins.h:369:23: anm: i expansionen av makrot "ORIG_Z_STEP_PIN"
 #define Z_STEP_PIN    ORIG_Z_STEP_PIN

                       ^
/tmp/arduino_build_147243/sketch/motion.h:673:12: anm: i expansionen av makrot "Z_STEP_PIN"
     WRITE( Z_STEP_PIN, LOW );
            ^
Using library SPI at version 1.0 in folder: /usr/local/arduino/hardware/arduino/avr/libraries/SPI 
Using library Wire at version 1.0 in folder: /usr/local/arduino/hardware/arduino/avr/libraries/Wire 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

RF1000, 2017-04-12:
Hello,

this version of the firmware must be compiled with Arduino V 1.6.5 (see https://github.com/RF1000/Repetier-Firmware/blob/master/README.md).

Best Regards,
    RF1000

GaramMasala, 2017-04-13:
Hello!

Yes I know, but that readme does not state it must have that exact version. Why is it compiling fine with 1.0.6 then?

mhier, 2017-04-13:
Obviously you don't need the exact version, I even managed to compile it with Arduino 1.8.2. Only it is not guaranteed to work with any other version than 1.6.5. The message is, if you have trouble you should try 1.6.5.

GaramMasala, 2017-04-15:
"On the other hand, one should try to fix the problems preventing the compilation with versions newer than 1.6.5..."

Yes, that's how it should be.

Nibbels, 2017-04-23:
Yes... "the community" fixed this ^^
but not this original firmware.

Have a look at mhier and my branch. you can even compile this version with cmake on raspberry pi.

GaramMasala, 2017-04-25:
Did find mhiers/branch profile but that was not much about repetier. Your profile/branch I did not find.

Nibbels, 2017-04-25:
Well you should have! One click on my Icon and then to https://github.com/RF1000community/Repetier-Firmware

zorgoban, 2022-06-04:
I'm still trying to find the spot in any of the sparse documentation where it tells me what board or processor to select for compilation. @Nibbels your version is of no use since it doesn't suppor SD-Card.