RF1000/Repetier-Firmware #23: Temperature Manager Sensor Error

fbeck55, 2020-11-06:
Installed the newest development build: 1.47.00 
right after the upload the temperature manager sensor error appears.
On the display all 3 heating zones are only diplayed as def/0
Tried to find out the cause by checking the pin.h. But the pins are same as before where it was still working. 

Reloaded build 1.45.02: no sensor errors all OK. I work with only one extruder, in build 1.45.02 I defined the
NUM_EXTRUDER = 1
in the new build 1.47.00 I tried both with NUM_EXTRUDER =1 and =2 
(=1 I needed to define some more variables (calibration position mm and steps) for the heatbed scan that where inside the "if NUM_EXTRUDER=2" brackets, and therefore not defined for 1 Extruder)

Is there any change in the way of checking the temperature manager?

mhier, 2020-11-09:
Are you sure it is not a loose cable? The temperature manager stays in the fault condition, if it has detected it once, until the firmware has restarted.

Also check carefully that the sensor types are correct. I think the error checking is done by checking whether the temperature is in a sensible range. If you have the wrong sensor type configured, this might be the case always.

I can only test with a single extruder, there everything works. In addition, my development effort has shifted to Klipper now, so I cannot really offer my support here any more...

fbeck55, 2020-11-10:
Hi mhier,
Danke für deine Antwort.

No loose cable, in 1.45.02 bed and ext0 are working both correctly. I reloaded the old version again to check.

Ext1 is not connected in my setup.
My first question therefore: is the new configurable left or right hotend checked or not when not configured? 

So do I need to set NUM_EXTRUDER to 1 to make sure ext1 is not checked? 
Or is NUM_EXTRUDER = 2 configured only the right one equivalent?

I tried to set it to 1 but there were not defined variables for RF_2000, which made me a little insecure if this was the right way... 
I defined them, copying what was defined for RF_1000. 

The result: only 2 heating zones were shown not 3 which is good, but still i have the temperatur manager sensor error.

I need to check the configuration of the temperature sensors that could be a reason...

mhier, 2020-11-10:
I cannot tell you this, since I have an RF1000 with a single extruder only... Maybe you can ask at the rf1000.de forum, there should be other people with a similar configuration (RF2000 single extruder and community firmware).

Nibbels, 2020-11-10:
Hello fbeck55, 

you talk about the 1.45.02 so you did use the community firmware https://github.com/RF1000community/Repetier-Firmware/tree/community_development , which is not the official firmware. 
Here you find a version 1.47, the manufacturers official firmware. 

The pure version number of the 1.47 version is higher than in the community mod branch, but the amount of features is smaller and possible pitfalls within the configuration are quite high.

For the community mod (1.45.02) you need to set RF2000 and num_extruders = 1. That is correct.
In case you have no second extruder and cannot solve the riddle attach some resistor to the empty ports. You can measure the other sensor when it is at room temperature. (Do I remember 83ohms ??)

Another trick might be to reset the EEPROM settings (change the eeprom version to another number https://github.com/RF1000/Repetier-Firmware/blob/0e52693eaed1f606af411e4898030b033304b94f/RF2000/Repetier/Configuration.h#L58). But that is just a random hint which might not work as expected. If nothing else works its worth a try.

fbeck55, 2020-11-13:
Hi Nibbles, thank you very much for your totally right conclusion that I took accidentally the Conrad supported version, which I did not wanted to use! They just have so similar names and versions. 
So for me this bug is obsolete even if it still seems to be one (at least that, you cannot define 1 Extruder without getting exceptions seems not to be correct for me...)

Therefore I close this topic. @mhier: Thanks also for your hint about Klipper and the forum, I will probably follow both advices.

closed 2020-11-13 by fbeck55